AndrewEllis93/Print-Tuning-Guide #9: Issues With SuperSlicer Profiles

Declan0811, 2022-02-04:
So, I just went throught this entire guide and everything went nicely, until I used the superslicer profiles. Whenever I slice with them, I get an error that Python 3.1 cannot be found in the C:/ directory, even though I have python installed. Though this error occurs, a gcode file still exports, albeit when trying to print it, the bed doesn't heat during the print, and the print usually stops during the start of the print. Any help in fixing this is greatly appreciated, thanks!

AndrewEllis93, 2022-02-04:
You need to specify the paths for your particular installation, my paths will likely be different from yours. 

Please seek assistance in the Voron Discord (#slicers_and_print_help channel), GitHub issues are meant more for an avenue to resolve issues with the guide itself, rather than troubleshooting assistance.

closed 2022-02-04 by AndrewEllis93

Declan0811, 2022-02-04:
> You need to specify the paths for your particular installation, my paths will likely be different from yours.
>
> Please seek assistance in the Voron Discord (#slicers_and_print_help channel), GitHub [...]

Ah ok, sorry for the bother! I'll go fix that :p