AndrewEllis93/Print-Tuning-Guide #10: Pressure advance and part mechanical strength

elcrni, 2022-02-12:
Hi there, not sure how else to comment/contribute, so apologies.

I am in a process of doing extensive testing and research on Voron 2.4 and what i've found with Pressure Advance is that it directly affects mechanical strength of the part (Logical!) and your amazing article suggest to always stay on the lower side of the PA value... this can and will mask the issue, the print will look nicer as there will be no gaps close to corners BUT, this will cause serious "perimeters fusing together" issues making the parts much weaker.

What i did as a test, i keep printing "[a]_xy_joint_cable_bridge_igus.stl" from Voron 2.4 repository as this part has everything, 90 degree walls, 45 degree walls, holes, bridges... and what i have found is if the EM is correct and PA too low, i can easily breakl the part with my fingers, when the PA is higher (in my case cca 0.06) for filamentum ASA i can not break the part no matter how hard i try. same setup, PA different. 

So yes, lower values do look nicer and more correct, but the part is mechanically weaker.

Thanks,
Alek

AndrewEllis93, 2022-02-12:
Do you use Discord? May he easier to discuss via DM / exchange photos etc 

[Ellis#4980[(
https://discordapp.com/users/207622442842062849)

AndrewEllis93, 2022-02-28:
Closing, discussed privately, multiple variables were being changed, unsure of root cause...

closed 2022-02-28 by AndrewEllis93