AndrewEllis93/Print-Tuning-Guide #12: TEST_SPEED macro considerations for CoreXZ machines (ie, Switchwire)

sloscotty, 2022-02-21:
Hi.  Finished tuning my V0.1 and moved on to my SW.

The TEST_SPEED macro reported wildly different positions for the X-axis upon completion of the test.  Scratched my head for awhile, then realized that the Z motor contributes to X motion.  If the Z beginning position is not returned to the position at the initial GET_POSITION state, then X position will be way off (more than 6000 microsteps in my case).

Not sure how to best handle this.  Simplest seems to do the GET_POSITION after a full home (G28) at beginning and end, but I don't know if that introduces other problems.  I may just try to undo the G0 ... Z{bound+10}...

In any case, it is probably a good idea to mention in the docs that for CoreXZ machines, the Z position values should be compared as well.

Thanks for the guide!  My V0.1 is purring like a tiger!

sloscotty, 2022-02-21:
I decided to only change the final G28 X Y to a full home G28.  It appears to work on switchwire as expected.  Left the remainder of the macro untouched. (I use [safe_z_home], so after homing my toolhead is in the center of the bed.)

AndrewEllis93, 2022-02-28:
Just made this change, thanks for letting me know

> I decided to only change the final G28 X Y to a full home G28. It appears to work on switchwire as expected. Left the remainder of the macro untouched. (I use [safe_z_home], so after homing my toolhead is in the center of the bed.)

closed 2022-02-28 by AndrewEllis93

mstawick, 2022-10-25:
Hey @AndrewEllis93 , 

FYI, it looks like the macro here:
https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/macros/TEST_SPEED.cfg

still uses the original G28 X Y instead of just G28. Found this issue while tuning my SW. 

Cheers, thanks for the great guide

AndrewEllis93, 2022-10-25:
@mstawick Did you have to change the middle G28 X Y to G28? The one with the comment about hall effect endstop accuracy

mstawick, 2022-10-25:
@AndrewEllis93 

I've changed the one in line 102:

```
# Re-home and get position again for comparison:
G28 X Y      <--- changed this to simply G28
# Go to XY home positions (in case your homing override leaves it elsewhere)
```

At the very beginning, the macro does G28 and then the initial get_position, then it moves Z to the height where the test is performed:
```
	# Go to starting position
	G0 X{x_min} Y{y_min} Z{bound + 10} F{speed*60}
```

so doing full G28 at the end puts it again in the correct height before doing the final get_position

AndrewEllis93, 2022-10-28:
Implemented in 4a287b097c0f34d0712fa66eb188d80a8a2f7edb.