AndrewEllis93/Print-Tuning-Guide #15: Input shaping: before PA

dewi-ny-je, 2022-03-22:
Wonderful guide.
Would you please edit it to make it clear that input shaping should be performed before PA, since even small amounts of ringing will affect the appearance of corners and lead to wrong estimates?
Thanks

dewi-ny-je, 2022-03-22:
Probably because you have a Voron and ringing is limited to begin with!
I noticed that the PA tower before shaping is more difficult to read, since both ringing and PA affect how corners look like.
Without shaping and with a printer less stuff than Vorons it's very easy to pick a PA value higher than it should be.

Also, input shaping with an accelerometer is not affected by anything else, since its a purely instrumental optimisation. PA and corner sharpness is affected by toolhead motion, therefore I find it makes sense to do shaping before PA.

AndrewEllis93, 2022-04-04:
Just added said note at the beginning of the section.

closed 2022-04-04 by AndrewEllis93