AndrewEllis93/Print-Tuning-Guide #20: FR: hope we can include in the pressure advance lines method a bed for delta printers

jpbnto, 2022-05-10:
hope there is an option to change the bed shape to circle for our delta printers

https://realdeuce.github.io/Voron/PA/pressure_advance.html

AndrewEllis93, 2022-05-13:
The straight lines should work on a delta just fine. Or do they not fit or something?

jpbnto, 2022-05-13:
> The straight lines should work on a delta just fine. Or do they not fit or something?

My delta have a small bed . I get out of range error in klipper because 0,0 is in the center of the bed unlike the square beds that 0,0 is on the corner.  

I tried to use the marlin line method but i get bad results.

[image]

On my corexy i have no problem with the lines method. :)

AndrewEllis93, 2022-05-17:
I see. I've never used a delta so forgive the ignorance.
I've dropped the maintainer a message, we'll see what he comes back with. 
For now, you can use the Marlin site with the old macro. See here for the older version of the directions:
<https://github.com/AndrewEllis93/Print-Tuning-Guide/tree/1cb1655ee0351136689da14e2bac67c39b2cb08c#method-1>

jpbnto, 2022-05-17:
> I see. I've never used a delta so forgive the ignorance. I've dropped the maintainer a message, we'll see what he comes back with. For now, you can use the Marlin site with the old macro. See here for the older version of the directions: https://github.com/AndrewEllis93/Print-Tuning-Guide/tree/1cb1655ee0351136689da14e2bac67c39b2cb08c#method-1 (may have to click the link twice for it to jump properly)

thanks, will wait and hopefully the creator updates it. 👍

constaac, 2022-07-25:
Did the maintainer respond about this? I think the issue is delta printers home at the center. I haven't tried, but I assume the coordinates can be tweaked with that in mind

AndrewEllis93, 2022-07-25:
I poked him again

AndrewEllis93, 2022-09-12:
And again! :)

AndrewEllis93, 2022-11-05:
@jpbnto @constaac 
I have added delta support back here, but I am unable to test it as I don't own a delta printer. If you would be so kind :)
https://andrewellis93.github.io/pressure_advance/pressure_advance.html

jpbnto, 2022-11-05:
Oh nice. Will report back to you once i test this. Thank you so much.

AndrewEllis93, 2022-11-07:
Glad to hear. Thanks for testing.  I will replace the guide link.

closed 2022-11-07 by AndrewEllis93