AndrewEllis93/Print-Tuning-Guide #21: Question: Is temperature calibration included in this guide?

Elec-theo, 2022-05-11:
Hi :)

I couldn't find temp tuning and I wondered what its relation would be to pressure advance and input shaper.

Thanks for putting this great guide together!!

Theo

AndrewEllis93, 2022-05-13:
Currently no. Honestly I have never mess with it too much, I generally just throw a temp at it and call it a day, and raise it if I need a bit more flow or strength. Maybe something in the future, though.

closed 2022-05-13 by AndrewEllis93

Elec-theo, 2022-05-14:
Alright, I use the temperature tower from Calibration Shapes (Cura plugin). Works really well :)

Cheers

Harrypulvirenti, 2022-12-12:
@AndrewEllis93 I was going to post a question exactly about the same topic expecially about the order of when to print the temperature tower.
Which is your suggestion? Before the PA/EM or after?

Maybe you can include this information in the guide.

Ps thanks for the amazing work

AndrewEllis93, 2022-12-12:
> @AndrewEllis93 I was going to post a question exactly about the same topic expecially about the order of when to print the temperature tower. Which is your suggestion? Before the PA/EM or after?
> 
> Maybe you can include this information in the guide.
> 
> Ps thanks for the amazing work

The other part of the issue is that temperature towers are only really useful for PLA, because it likes cooling and doesn't mind short layer times.

With pretty much any other material, you have to crank the fan very high to stop the towers from being completely melty, which in turn weakens the print, and in turn completely invalidates the results. 

Most of these other materials just don't work well with thin objects and short layer times, which is exactly what a temp tower is, making it unrealistic. 

Adding to this, I have truly just never found the need. Most filaments are perfectly happy in a pretty wide range of temps, you really can just throw a known number at it and it's fine 98% of the time. If it's a bit weak, give it a bit more. That's really all there is to it, in my mind.

The closest reasonable thing would be to print a bunch of objects at the same time (thereby introducing reasonable layer time), setting a different temp for each in your slicer, and then breaking them with some channel locks or something to compare strength.

Harrypulvirenti, 2022-12-12:
> > @AndrewEllis93 I was going to post a question exactly about the same topic expecially about the order of when to print the temperature tower. Which is your suggestion? Before the PA/EM or after?
> > Maybe you can include this information in the guide.
> > Ps thanks for the amazing work
> 
> The other part of the issue is that temperature towers are only really (sort of) useful for PLA, because it likes cooling and doesn't mind short layer times. PLA is frankly very weird/atypical in this way.
> 
> With pretty much any other material, you have to crank the fan very high to stop the towers from being completely melty, which in turn weakens the print, and in turn completely invalidates the results.
> 
> Most of these other materials just don't work well with thin objects and short layer times, which is exactly what a temp tower is, making it unrealistic. They would mislead more than help.
> 
> Adding to this, I have truly just never found the need. Most filaments are perfectly happy in a pretty wide range of temps, you really can just throw a known number at it and it's fine 98% of the time. PLA included. If it's a bit weak, give it a bit more. That's really all there is to it, in my mind.
> 
> The closest reasonable thing would be to print a bunch of objects at the same time (thereby introducing reasonable layer time), setting a different temp for each in your slicer, and then breaking them with some channel locks or something to compare strength.

Thanks for the quick answer!
This clarify a lot of doubts.