AndrewEllis93/Print-Tuning-Guide #24: TEST_SPEED macro accuracy issue with hall-effect endstop

linyongzheng, 2022-05-27:
If the carriage is too close to the hall effect endstop for the first homing, the microsteps can potentially be miscounted during the G28 sequence.

I remedied this by having the macro move to the center of the bed before the initial G28. Might want to mention this potential issue in the work instructions or modify the macro to account for hall-effect endstop users.

AndrewEllis93, 2022-06-19:
How do you mitigate this? I don't have a hall effect endstop, so I'm not too familiar with it. 

For the first homing sequence, would it be best to conditional home, move away 2cm or so, home again for the actual measurement, then report position?

AndrewEllis93, 2022-06-20:
In the meantime, I added a note about hall effects endstops.

linyongzheng, 2022-06-21:
I've been doing a conditional home, then move to center of bed before
homing to report position. I'm sure there are better ways to do this but it
has been working for me.

AndrewEllis93, 2022-08-09:
I have modified the macro, can you please see if that helps and let me know?

AndrewEllis93, 2022-09-08:
Modified the macro to move away before first measurement homing

closed 2022-09-08 by AndrewEllis93