AndrewEllis93/Print-Tuning-Guide #28: Twisted/skewed frame

dewi-ny-je, 2022-07-06:
Based on 
https://discord.com/channels/431557959978450984/582347488938164240/993958021283725433
and on the answer
https://discord.com/channels/431557959978450984/582347488938164240/994063694436782201
which suggests a skewed/twisted frame, would it be possible to extend the page about gantry squaredness to a more general squaredness of the outer frame (even vertically), with tips to check and correct?
Of course it's less relevant for Vorons, since they have 4 point bed screws and a gantry independent from the outer frame, but your guide is very good and used also by people with different printers!

In the case of that post, the author with the issue has a glass bed so that can be used as reference, but given the photo posted, the gantry is using 8 mm rods so it will sag, making even a perfect frame show some bowing.

It's tricky, but maybe with your experience you can extend the guide by providing at least tips.

AndrewEllis93, 2022-08-09:
Sorry been procrastinating this one purely because I don't know how exactly to approach it. There are a lot of different printers out there, so it's hard to give generalized advice for this.

dewi-ny-je, 2022-08-10:
> Sorry been procrastinating this one purely because I don't know how exactly to approach it. There are a lot of different printers out there, so it's hard to give generalized advice for this.

I understand.
Maybe just mentioning that a mesh like the one I provided may be indicative of a skewed frame could be enough to make people aware so that they can check.

closed 2022-08-10 by dewi-ny-je