AndrewEllis93/Print-Tuning-Guide #30: Can you please add some guide for the labeled extrusion multiplier cubes?

samuellazea, 2022-08-08:
Hello,

I see in the test prints folder that you have some labeled cubes and I like the idea to have them visible even after you mix them ( cause if you print multiple normal cubes you can start to mix them) but I would like if possible to have a section in the guide  on how to use them. For now I don't have an idea what extrusion multipliers to set on each one to be able to test it properly.
Thank you @AndrewEllis93

millerlarson, 2022-08-08:
> but I would like if possible to have a section in the guide on how to use them. For now I don't have an idea what extrusion multipliers to set on each one to be able to test it properly.

Does the [section about EM](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/extrusion_multiplier.md#method) not cover this thoroughly enough?
If not, where exactly are you getting stuck or what is unclear?

AndrewEllis93, 2022-08-08:
I'm not entirely clear on what the confusion is either. You would follow the normal EM tuning instructions, and if you want you can use the labeled cubes for each step rather than the blank ones.

closed 2022-09-08 by AndrewEllis93