AndrewEllis93/Print-Tuning-Guide #33: Commit description

dewi-ny-je, 2022-09-08:
Hi, thanks for your work first of all.
I followed the guide and I got your PIF profile and from time to time I come back to check what changes/improvements are there.

I find however very time consuming to open each single commit to find out, all of them are called "update" with, usually, nothing else.

I think that giving more meaningful titles and a line or two of description would help the (several) users of your guide and profiles to follow better!
I understand it would take you a bit more time for each commit so you may not want to follow the suggestion, but I thought it was worth to mention it to you so you are aware of it.

AndrewEllis93, 2022-09-08:
Just a bad habit really. I commit too often so much of the time it's just ten typo fixes in a row. I think I'm done, then I find another little thing, rinse, repeat. So, maybe 😅

closed 2022-09-08 by AndrewEllis93

AndrewEllis93, 2022-09-13:
Welp, breaking the habit so far! Probably about time :)
Maybe a bit wordy with the commit messages, but I'm getting there, hah.