AndrewEllis93/Print-Tuning-Guide #37: Klipper resume after power loss

LuckWizard, 2022-10-10:


conlank, 2022-10-11:
As far as I know Resume after power loss is not supported by Klipper. For a more in depth read about the reasons why, please read this [thread].(https://klipper.discourse.group/t/resume-print-after-power-loss/2682)

AndrewEllis93, 2022-10-11:
Resume on power loss is iffy regardless, by far the better option is to buy a UPS. It only really works with the simplest of printers (i3)

Much of the time your print will release when it cools down. Even when you manage to resume, you'll often get a slight misalignment at best, but with larger printers thermal expansion means your z height will likely be off on resume.

And redoing QGL is impossible with a print in the way, likely same story with other auto leveling methods.

My UPS has saved a few prints for me. 

The other option is to measure where it stopped, and just cut the STL up to print the other half & glue it together.

closed 2022-10-11 by AndrewEllis93