AndrewEllis93/Print-Tuning-Guide #39: Example voron2 klipper configs missing

evilphish, 2022-10-26:
Hi Andrew,

I have recently been building my Voron 2 and your "backup klipper configs" were a huge help for me. I just wanted to look up your thermistor setup and noticed they were gone. Any chance you could upload them again, maybe add them to the print tuning guide next to "my setup"? Those were really helpful to me!

Thank you for all your articles and writeups. Your mentioning of belt slack on the individual z-stepper units actually made me fix the probing issues I have had :)

best
Alex

AndrewEllis93, 2022-10-26:
Hi, I made that repo private because it was originally just my own backup, it wasn't really intended for sharing. 

There are some "quick and dirty" things in there - mostly macros - that were spreading around the community more than I'd like. And some people were just attempting to just use my configs as-is.

When it was smaller scale, it wasn't an issue, but as the guide has grown in popularity, more and more people were stumbling into my personal configs and taking things as gospel.

I have moved sanitized/documented versions of some of my macros to the guide. As for the rest of the config, I'm just trying to think if there's a way to share it but keep the macros part hidden... I'm not quite a git wizard

EricZimmerman, 2022-10-26:
you could use a .gitignore file to leave OUT the macros folder.

put it in your macros folder, then add this to it:

```
*
*/
!.gitignore
```

AndrewEllis93, 2022-10-26:
Yes, but it's a backup in case my printer explodes, that would just cause it not to back up

evilphish, 2022-10-26:
I was actually reading through your macros, learning a lot and incorporating some ideas into my own macros. This was very fruitful for me. In general, going through the different cfgs and looking up how you tied things together was super helpful. Maybe, as long as there isn't anything personal in your macros that you do not want shared in general, you could sort of make the files available with a big disclaimer for people not to use them as is? 

Or maybe do it similarly to the guy that did the online configurator for the pressure advance calibration and include a certain macro or gcode that will prevent anyone from just copy/pasting your files? Essentially just turn off all motors, heaters and whatnot?  However, that would of course mean extra work on your part :/

Or use the Darwin approach to separate those who blindly copy and paste stuff from the internet from those with printers that still work :D

AndrewEllis93, 2022-10-27:
Well, here it is while I mull it over
[v2.247_backup_klipper_config-master.zip](https://github.com/AndrewEllis93/Print-Tuning-Guide/files/9876250/v2.247_backup_klipper_config-master.zip)

leesafur, 2022-10-28:
Thanks Andrew! I often refer to your setup too and have learnt a huge
amount from you... Couldn't have done it without you. lol

AndrewEllis93, 2022-10-28:
I made it public again. 
I may just fix up some of those "quick and dirty" macros eventually.

closed 2022-10-28 by AndrewEllis93

evilphish, 2022-10-28:
Thank you and also thank you again for all the effort you put into sharing your knowledge with others!