AndrewEllis93/Print-Tuning-Guide #40: TEST_SPEED macro on Delta printers

PerLangkjaer, 2022-11-04:
Hi
I found this via the Vector 3D channel on YT and really love your work..
So much so that all 5 of my klipper printers have been worked on today.
But....
The TEST_SPEED macro do no work on the Delta printer, I have been in the macro and tried to set it up, but its beyond my understanding unfortunatly.
What should be considered when testing on a Delta printer???

AndrewEllis93, 2022-11-05:
Hi, I don't have a delta so I can't really do much testing myself. That said, I tried to modify this to work with a center origin / round plate. See attached. 

Since I can't test this myself, I take no responsibility for any damage! (just a disclaimer, I don't actually expect any, but you never know...)

[TEST_SPEED_DELTA.txt](https://github.com/AndrewEllis93/Print-Tuning-Guide/files/9942745/TEST_SPEED_DELTA.txt)

Also, just for my own future reference, can you please give me the output of `DUMP_VARIABLES` on your delta? Macro can be found here:
https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/main/articles/useful_macros.md#dump-variables

AndrewEllis93, 2022-11-05:
@PerLangkjaer I have added delta support back here, but I am unable to test it as I don't own a delta printer. If you would be so kind :)
https://andrewellis93.github.io/pressure_advance/pressure_advance.html

PerLangkjaer, 2022-11-05:
@AndrewEllis93 I’ll try it out as soon as I have the time. Possible Sunday otherwise Monday

AndrewEllis93, 2022-11-05:
@PerLangkjaer That second comment of mine was meant for the other github issue, mixed them up! Only need you to test the macro dump variables bits, but you're more than welcome to try that PA test too if you're feeling spicy

AndrewEllis93, 2022-11-05:
One thing just occurred to me. Currently it runs the test at height Z30. If one of the motors were to skip, could that result in a bed collision?

You may want to change the Z height in this block (line 59) to something higher, if that's the case:
```
# Go to starting position
G0 X{x_min} Y{y_min} Z{bound + 10} F{speed*60}
```

AndrewEllis93, 2022-11-05:
This version will go to maximum Z minus bound (default -20mm) instead of Z30.
[TEST_SPEED_DELTA.txt](https://github.com/AndrewEllis93/Print-Tuning-Guide/files/9944497/TEST_SPEED_DELTA.txt)

jpbnto, 2022-11-06:
i can help you out with this test_speed macro

try this
[gcode_macro TEST_SPEED]
gcode:
	# Speed
	{% set speed  = params.SPEED|default(printer.configfile.settings.printer.max_velocity)|int %}
	# Iterations
	{% set iterations = params.ITERATIONS|default(5)|int %}
	# Acceleration
	{% set accel  = params.ACCEL|default(printer.configfile.settings.printer.max_accel)|int %}
	# Bounding inset for large pattern (helps prevent slamming the toolhead into the sides after small skips, and helps to account for machines with imperfectly set dimensions)
	{% set bound = params.BOUND|default(40)|int %}
	# Size for small pattern box
	{% set smallpatternsize = SMALLPATTERNSIZE|default(20)|int %}
	
	# Large pattern
		# Max positions, inset by BOUND
		{% set x_min = printer.toolhead.axis_minimum.x + bound %}
		{% set x_max = printer.toolhead.axis_maximum.x - bound %}
		{% set y_min = printer.toolhead.axis_minimum.y + bound %}
		{% set y_max = printer.toolhead.axis_maximum.y - bound %}
	
	# Small pattern at center
		# Find X/Y center point
		{% set x_center = (printer.toolhead.axis_minimum.x|float + printer.toolhead.axis_maximum.x|float ) / 2 %}
		{% set y_center = (printer.toolhead.axis_minimum.y|float + printer.toolhead.axis_maximum.y|float ) / 2 %}
		
		# Set small pattern box around center point
		{% set x_center_min = x_center - (smallpatternsize/2) %}
		{% set x_center_max = x_center + (smallpatternsize/2) %}
		{% set y_center_min = y_center - (smallpatternsize/2) %}
		{% set y_center_max = y_center + (smallpatternsize/2) %}

	# Save current gcode state (absolute/relative, etc)
	SAVE_GCODE_STATE NAME=TEST_SPEED
	
	# Output parameters to g-code terminal
	{ action_respond_info("TEST_SPEED: starting %d iterations at speed %d, accel %d" % (iterations, speed, accel)) }
	
	# Home and get position for comparison later:
		G28
		# QGL if not already QGLd (only if QGL section exists in config)
#		{% if printer.configfile.settings.quad_gantry_level %}
#			{% if printer.quad_gantry_level.applied == False %}
#				QUAD_GANTRY_LEVEL
#				G28 Z
	#		{% endif %}
		#{% endif %}	
		# Move 50mm away from max position and home again (to help with hall effect endstop accuracy - https://github.com/AndrewEllis93/Print-Tuning-Guide/issues/24)
		#G90
		#G1 X{printer.toolhead.axis_maximum.x-50} Y{printer.toolhead.axis_maximum.y-50} F{30*60}
		#G28 X Y
		#G0 X{printer.toolhead.axis_maximum.x-1} Y{printer.toolhead.axis_maximum.y-1} F{30*60}
		#G4 P1000 
		GET_POSITION

	# Go to starting position
	G0 X{x_min} Y{y_min} Z20 F{speed*60}

	# Set new limits
	SET_VELOCITY_LIMIT VELOCITY={speed} ACCEL={accel} ACCEL_TO_DECEL={accel / 2}

	{% for i in range(iterations) %}
		# Large pattern
			# Diagonals
			G0 X{x_min} Y{y_min} F{speed*60}
			G0 X{x_max} Y{y_max} F{speed*60}
			G0 X{x_min} Y{y_min} F{speed*60}
			G0 X{x_max} Y{y_min} F{speed*60}
			G0 X{x_min} Y{y_max} F{speed*60}
			G0 X{x_max} Y{y_min} F{speed*60}
			
			# Box
			G0 X{x_min} Y{y_min} F{speed*60}
			G0 X{x_min} Y{y_max} F{speed*60}
			G0 X{x_max} Y{y_max} F{speed*60}
			G0 X{x_max} Y{y_min} F{speed*60}
		
		# Small pattern
			# Small diagonals 
			G0 X{x_center_min} Y{y_center_min} F{speed*60}
			G0 X{x_center_max} Y{y_center_max} F{speed*60}
			G0 X{x_center_min} Y{y_center_min} F{speed*60}
			G0 X{x_center_max} Y{y_center_min} F{speed*60}
			G0 X{x_center_min} Y{y_center_max} F{speed*60}
			G0 X{x_center_max} Y{y_center_min} F{speed*60}
			
			# Small box
			G0 X{x_center_min} Y{y_center_min} F{speed*60}
			G0 X{x_center_min} Y{y_center_max} F{speed*60}
			G0 X{x_center_max} Y{y_center_max} F{speed*60}
			G0 X{x_center_max} Y{y_center_min} F{speed*60}
	{% endfor %}

	# Restore max speed/accel/accel_to_decel to their configured values
	SET_VELOCITY_LIMIT VELOCITY={printer.configfile.settings.printer.max_velocity} ACCEL={printer.configfile.settings.printer.max_accel} ACCEL_TO_DECEL={printer.configfile.settings.printer.max_accel_to_decel} 

	# Re-home and get position again for comparison:
		G28 #X Y
		# Go to XY home positions (in case your homing override leaves it elsewhere)
		#G90
		#G0 X{printer.toolhead.axis_maximum.x-1} Y{printer.toolhead.axis_maximum.y-1} F{30*60}
		G4 P1000 
		GET_POSITION

	# Restore previous gcode state (absolute/relative, etc)
	RESTORE_GCODE_STATE NAME=TEST_SPEED

jpbnto, 2022-11-07:
i test> One thing just occurred to me. Currently it runs the test at height Z30. If one of the motors were to skip, could that result in a bed collision?
> 
> You may want to change the Z height in this block (line 59) to something higher, if that's the case:
> 
> ```
> # Go to starting position
> G0 X{x_min} Y{y_min} Z{bound + 10} F{speed*60}
> ```

i tested the TEST_SPEED_DELTA.cfg

edited this part because it gives an out or range error

```
  # Re-home and get position again for comparison:
        G28
        # Go to XY home positions (in case your homing override leaves it elsewhere)
        G90
        **#G0 X{printer.toolhead.axis_maximum.x-1} Y{printer.toolhead.axis_maximum.y-1} F{30*60}**
        G4 P1000 
        GET_POSITION
```

AndrewEllis93, 2022-11-07:
Ah missed that. Apart from that line, did it work? That line may just be removed.

PerLangkjaer, 2022-11-07:
[TEST_SPEED_DELTA.txt](https://github.com/AndrewEllis93/Print-Tuning-Guide/files/9949431/TEST_SPEED_DELTA.txt)
So this works great :-D

jpbnto, 2022-11-07:
> Ah missed that. Apart from that line, did it work? That line may just be removed.

yes its working.

AndrewEllis93, 2022-11-07:
I have put the file with that edit in the macros folder (and removed some of the unnecessary stuff).

PerLangkjaer, 2022-11-07:
Cool :-)
I'm cloesing it as it's fixed :-D

closed 2022-11-07 by PerLangkjaer

jpbnto, 2022-11-08:
andrew you forgot this part
```
  # Go to starting position
    G0 X{x_min} Y{y_min} Z{printer.toolhead.axis_maximum.z - bound} F{speed*60}
```
change to this    
```
    # Go to starting position
G0 X{x_min} Y{y_min} Z{bound + 10} F{speed*60}
```

AndrewEllis93, 2022-11-08:
Isn't it safer on Delta to run this further from the build plate? What issue is that causing?

jpbnto, 2022-11-08:
When using this line

`G0 X{x_min} Y{y_min} Z{printer.toolhead.axis_maximum.z - bound} F{speed*60}`

It gives an out or range error because one arm cannot go higher than the endstop. 

We can also use this 

```
  # Go to starting position
    G0 X{x_min} Y{y_min} Z{printer.toolhead.axis_maximum.z / 2} F{speed*60}

```
To be on the safe side, printhead between endstops and bed.

AndrewEllis93, 2022-11-08:
I see. I had assumed that the max Z height on a delta would be the maximum height with full range of motion rather than the max at center. 

I will change this later today.

AndrewEllis93, 2022-11-08:
Changed to `G0 X{x_min} Y{y_min} Z{printer.toolhead.axis_maximum.z/2} F{speed*60}` in 0283b1549a8a23f6504e988101879f683370d8e3