AndrewEllis93/Print-Tuning-Guide #45: The best way to calibrate pressure advance!

Tapap84, 2022-12-02:
I've just stumbled across this from a 3 month old reddit post. I've always had issues getting my pressure advance right until I found your tool. 5 minutes and my problems are gone. Really easy to read the results. The website is easy to use. What worries me is how long will it be hosted. Would be epic if this able to be downloaded and ran offline.

AndrewEllis93, 2022-12-02:
Glad it helped! I'm extremely pleased with how accurate it's turning out to be for everyone so far!

You can run it offline. The repo for it is here: https://github.com/AndrewEllis93/Pressure_Linear_Advance_Tool

If you download it, you should be able to just double-click on index.html and it should work!

AndrewEllis93, 2022-12-02:
Side note, it's hosted on GitHub Pages, not costing me anything, so probably not going away any time soon

closed 2022-12-02 by AndrewEllis93

Tapap84, 2022-12-02:
I see, thank you.

I'm reading through more of your stuff and this is amazing. I wish I found
this when I started. If you don't mind I'd like to link people here when
the flood of Christmas people start asking questions

AndrewEllis93, 2022-12-02:
Of course. Get the word out!