AndrewEllis93/Print-Tuning-Guide #49: Instructions for 24V fans using tachometer pin don't work well

Borfonius, 2022-12-21:
Simply plugging the tachometer pin into a free endstop port doesn't work well.  The pullup voltage on the endstop pins is too low compared to the 24V power to the fan, so the fan spins constantly, sinking current through the tachometer.

My suggestion to fix this is to add the following:
- BAT85 Schottky diode facing away from the MCU (just like with the inductive probe)
- 10k resistor pullup to 24V on the fan side of the BAT85.  I used one of the available always on FAN ports to get the 24V.

This solved the issue for me.  It wasn't particularly pretty having all of those components along the tachometer wire, but it worked.  The fan behaved normally, and the macro to detect fan stoppage worked as well.

AndrewEllis93, 2022-12-21:
This has come up a couple times in the Discord but so far nobody had given me a concise explanation for the guide - so thank you
Does this apply to all fan models? Mine didn't need this, and from what I've been told, different fans handle tach differently. If it's different for different tach setups, may need to account for that too?

Borfonius, 2022-12-21:
Hi Andrew,

I only tried this with one 24V fan.  The Orion OD4010-24HB01A that you
recommended in your guide:

http://d.digikey.com/dc/mn-w0iJh4uEE_bUitNCuXsaMqkw5UctQEQXQivOAnBMAxAqa2HiOed3QHpulfrGC9uqjsa9263lx8xAQBJoeKPKVJN2rb5EeRW-l-F7RFPpNkcpnD2mAIasy2ZRJXF7NUL4LNFrcHvDfycUY-WTQi-jYNWN-Ag2BoQNlTZZZYmP3aig-a0oGMztYufJUamekU74MAz42m2CDHE9AXscc14CsdQbzmFZu4HDOaWSiTKriWYR8DvqkKud0yJpSEDyK/MDI4LVNYSy01MDcAAAGIslrFr5ayOn3Q9Q9rAQH7wibQvG3CdS4E1yENALMy1Lzr160zjiPAMU68Z4zHpvOmQWbkoNw=

The datasheet has the following diagram which made me think that the board
pullup voltage was too low:

[image: image.png]

Though it shouldn't hurt to do this modification for any of the fans,
right?  It is definitely messy looking without a PCB though!

Let me know if you need more info!

Best Regards,

Christian

allenrowand, 2023-01-15:
Can confirm the issue- did this with the OD4010-24HB01A, got the always on fan and burned out my SB2040 (since RP2040 SOCs are very sensitive to overvoltage). Installed the BAT85 as described above, mod works as expected with the EBB36. I didn't need to add a physical pullup resistor, just did it in software.

Gil80, 2023-03-21:
Hi all,

My MCU is SKR mini E3 V3. I was following the PWM fan monitoring guide.

I ran into an issue where once I connect the 3rd wire (techometer) the fan will get a constant 20.4v making it hot to the touch and smelly. This voltage is supplied even if the hotend is cold.

This is the board diagram: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V3.0/Hardware/BTT%20E3%20SKR%20MINI%20V3.0_PIN.pdf

I connected the 2 pins to the FAN 1 header and the 3rd PWM wire to the Z-endstop PC2 pin.

It seems that no matter which PWM pin I choose, I either get `ADC error min/max temp out of range` or this constant 20.4v to the fan while it's not spinning.

Is there any other way I can kill the heaters if there's a fan failure of the temp near the heatbreak is rising?

Gil80, 2023-03-21:
> Simply plugging the tachometer pin into a free endstop port doesn't work well. The pullup voltage on the endstop pins is too low compared to the 24V power to the fan, so the fan spins constantly, [...]

Hi. I just read your comment and I ran into the issue of not being able to utilize the PWM on a 2-pin Z-endstop on my SKR mini E3 V3. 
Do you happen to know if a 1N4004 diode will do the job as well?
And by "facing away" do you refer to the Cathode facing away?

Borfonius, 2023-03-22:
Here is the link to the voronVoron documentation on how the BAT85 diode is wired for the inductive probe:

https://docs.vorondesign.com/build/electrical/#bat85-diode

I did the same thing for the tachometer pin.  But I also added a 10k pull-up resistor since I don’t believe the fan I used has one internally like the inductive probe.

 I can’t say whether you could use a different type of diode for this purpose, but I wouldn’t recommend it, since the BAT85 is selected specifically as a protection device.

Gil80, 2023-03-22:
@Borfonius  is this diagram correct?
https://imgur.com/a/fhP1GQd

Borfonius, 2023-03-22:
It’s not quite right.   The BAT85 looks good, but you want the right hand side of your 10k resistor to be connected to your 12V supply instead of in series with the tachometer wire.
Right now, the way you have it, the resistor isn’t functioning as a pull-up.

Gil80, 2023-03-22:
@Borfonius ok I think I got it now https://imgur.com/a/twfKXCb
p.s. - I use 24v fan, that fan image is just something I found online.

Gil80, 2023-03-23:
@Borfonius thanks for the input. I fixed the flipped diode - thanks for noticing.
Here is what I believe is the correct diagram thanks to your kind help: https://imgur.com/a/zdKlLY0

Borfonius, 2023-03-23:
Yep you got it!
Thanks for drawing this up!

arkeet, 2023-11-10:
Bumping this - someone I was helping was having a similar issue with a 24V 3 wire fan, where the fan was always spinning. Having a diode in line with the tach wire (cathode towards fan) is important, as otherwise the controller board could draw current through this pin, potentially damaging the fan as well as itself, especially if there is no series resistor on the endstop pin. (The current would flow inside the fan from V- to tach through a protection diode, which is not meant to carry significant current.)

The diode is probably optional for a 5V fan, but I would say it is required for any higher voltage.