AndrewEllis93/Print-Tuning-Guide #50: Bulging caused by Arachne

Clon1998, 2022-12-21:
Hey Ellis,
I recently tried to troubleshoot a print of mine since it always had two layers that bulged.
I tried some of the methods you suggest but nothing seemed to help.
A closer inspection of the part in the slicer finally lead me to the solution.
The bulging was caused by the Arachne slicer settings in SS. Therefore I think it might be worth adding a note about inspecting the outer perimeters in the slicer using SS/PS Width view since it highlights the issue quite clearly.

With Arachne, layer width view:
[image]

Classic width, layer width view:
[image]

Clon1998, 2022-12-21:
Here is my 3mf file (Zipped due to Github)
[roland.zip](https://github.com/AndrewEllis93/Print-Tuning-Guide/files/10277074/roland.zip)

AndrewEllis93, 2022-12-21:
Thanks - do you have a photo of the print itself?

Clon1998, 2022-12-21:
yes of course.
My iphone won't focus any closer sadly but i tried my best to capture it.

With arachne:

[image]

With classic mode

[image]

oliof, 2023-01-13:
Is this specific to the PrusaSlicer Arachne implementation, or does it also happen in recent Cura releases (5.2 or newer) where Arachne is the default engine?