AndrewEllis93/Print-Tuning-Guide #53: Dynamic acceleration / square corner velocity (jerk)

Henk72, 2023-01-09:
Hi Andres,

I was missing this part in your tuning guide, so thought to mention it.

With SuperSlicer there is the possibility to set accelerations per extrusion type (e.g. external perimeter, infill, travel).
The square corner velocity (SQV)/jerk can not be set this way in the slicer. It can be done with a post-processing script, or with slicer g-code substitution or in the printer firmware. I used the latter and made an override of the M204 macro in Klipper. It deduces both 'acceleration to deceleration' and SQV form the requested acceleration by the slicer (M204).

What do you think about using high accelerations and SQV only for travel?

LAP87, 2023-01-12:
In Klipper we set up SQV in the printer.cfg, and using Fluidd/Mainsail you have the ability to change it on the fly.
I'm guessing you're looking at it from the marlin perspective?

Henk72, 2023-01-12:
I sometimes use it like you mentioned. However, If you make an on the fly adjustment in the Fluidd/Mainsail it will be applied for everything. So all axis and all extrusion types. The slicer knows the extrusion type and can change the acceleration and jerk on the fly. 
My question is if it will give some printing benefit (speed/quality). For instance low acceleration/decel/SQV for first layer. Very high for infill, and lower for perimeter. For travel even higher.