AndrewEllis93/Print-Tuning-Guide #62: TEST_SPEED crashes nozzle into bed after run completes.

bassamanator, 2023-02-09:
On my new printer, Sovol SV06, I did a run and after the routine, the printhead decided to crash into the bed. I believe after the run, there's a `z home` and that's when it just crashed. My probe is working fine.

Here's my [printer.cfg](https://github.com/bassamanator/Sovol-SV06-firmware/blob/a7d640bf1d6670d66894b4644a67e06af8e6ec16/printer.cfg).

PS I've used `TEST_SPEED` extensively on my Voron 0.1 and it works beautifully!

AndrewEllis93, 2023-06-13:
Sorry just returning from hiatus here. Still an issue?

bassamanator, 2023-06-13:
Yeah still is. The last time I tried it, the printhead crashed into the right pillar of my i3 style printer, instead of crashing the nozzle into the printbed. I've gone over the `TEST_SPEED` macro line by line and don't see any issues there. And again, on my V0.1, the macro works just fine.

Love your work btw, I point to https://ellis3dp.com/Print-Tuning-Guide/ constantly and consider your SS print profiles the best in existence!

closed 2023-07-26 by bassamanator