AndrewEllis93/Print-Tuning-Guide #63: Question for PA Test

xenon2008, 2023-02-12:
Hello together =)

@AndrewEllis93 
First of all, thank you for your wonderful work, I really take my hat off to that - it's unbelievable how much knowledge and time you put into it!

But could you tell me what exactly you mean by that:
_(And then I’d just cover up the slight gapping with single top perimeters - which is unfortunately exclusive to SuperSlicer.)_
[](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html)
Because I can't find this function in superslicer, or I'm just "just" looking for the wrong function or the wrong name

Kindly Regards from Upper Austria =)

millerlarson, 2023-02-12:
Have a look here:
[image]

xenon2008, 2023-02-13:
Ahhh there is it! Thank you so much for this fast Answer ;)  
  
KIndly Regards from Upper Austria

closed 2023-02-13 by xenon2008