AndrewEllis93/Print-Tuning-Guide #64: New Z-Offset Calibration Method

alexrmay91, 2023-02-25:
I've recently started tuning my z-offset in a new way I haven't seen before. I just want to share with you in case you like the method and want to add it to your guide.

I print a series of rectangles (1 layer high) side by side and change the infill to concentric.

[image]

[image]

If you're too close to the bed, you'll get "extra" plastic squishing out to the side and most of it will squish out towards the side with no plastic already on the bed. Since it's set to concentric infill, all that "extra" plastic will pool towards the middle, exaggerating the problem, making it really easy to see and tune. Just shine some light from the side while printing or afterwards.

Shown below, left to right, is an offset of: 0mm, -0.01mm, and -0.005mm. I tend to choose the lowest offset that has no bump, so in this case I adjusted my offset by -0.005mm. 

[image]

[image]

I saved an .slt to my shape gallery with 7 rectangles at 0.2mm height. Very easy to drop it in an adjust as needed.

BuddyBing, 2023-02-26:
Incredibly interesting, but I also wonder how much flow rate/PA would impact this?

alexrmay91, 2023-02-26:
I think PA would only affect the corners and the long straight section/middle would be unaffected.

As far as flow rate, that should be done before z-offset. Having an incorrect flow rate will mess up any visual-based z-offset calibration. High flow rate and too close to the bed look the same on the first layer, so you need to eliminate flow rate first.