AndrewEllis93/Print-Tuning-Guide #74: EM Calibration with PrusaSlicer

jleutgeb1, 2023-05-04:
In the Extrusion Multiplier Section for PrusaSlicer it is stated that "There is no way to set the EM per object. You will have to print the test objects one at a time." 

With the [PrusaSlicer Macro Language](https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-Macro-Language) introduced in 2.6.0-alpha6 there is now a way.

In "Printer Settings">"Custom G-code" one can add the "Start G-code":

{
global flowtestrate = 100;
"M221 S" + flowtestrate;
}

and under "Between objects G-code (for sequential printing)":

{
flowtestrate = flowtestrate - 2 ;
"M221 S" + flowtestrate ;
}

This will decrease the flowrate by 2 for each object when "Sequential printing" is enabled under "Print Settings">"Output options". It might also be useful for other calibrations where there is no built-in modifier.

I wanted to share this info if maybe someone else finds it useful. If there is an easier way, please let me know.

myevo8u, 2023-06-01:
I also created a post processing script for this yesterday for this as well! https://github.com/myevo8u/Prusa-Slicer-Extrusion-Multiplier-Calibration-Script

dotrey, 2023-08-05:
Based on @myevo8u's script (which I couldn't get to run), I used the  G-code substitutions in PrusaSlicer 2.6:
- add multiple objects (not instances) to your plate and name them `em-100`, `em-99.5`, `em-99` etc. 
- make sure both `Label objects` and `Complete individual objects` are enabled
- add a new G-code substitution with a regular expression `^(; printing object em-([0-9.]+).*)$` and replace `${1}\nM221 S${2}`
- this will add a `M221 S100` for the object `em-100`, a `M221 S99.5` for the object `em-99.5`, and so on

This method has the downside(?) that the `M221` command is added after every layer change, because the object label is added after every layer change. This is more often than necessary, but as far as I can tell it shouldn't cause issues.

Also, the regex could be improved to only work with valid numbers, as the current one will work with something like `em-0.1.2` and add a `M221 S0.1.2`.

I prefer this way over the `Between objects G-code (for sequential printing)` of @jleutgeb1 or the version in #75 because I can easily define which object will be printed with a certain extrusion multiplier, and can also vary freely which extrusion multipliers I want to test (e.g. sth like 100%, 99%, 98.7%, 98.235%, 95% all on the same plate).

hiltonjanfield, 2023-08-07:
> Based on @myevo8u's script (which I couldn't get to run), I used the G-code substitutions in PrusaSlicer 2.6:

This was an interesting solution I hadn't thought of, which I might start using because it isn't tied to printer settings - therefore I can use the same 3mf file for multiple printers without any manual editing. The only thing I don't like about it is that it requires manually renaming all the objects rather than changing a variable or two. One enhancement I might suggest is to use the filename format from Ellis' [test_prints/labeled objects folder](https://github.com/AndrewEllis93/Print-Tuning-Guide/tree/main/test_prints/extrusion_multiplier_cubes/labeled) (`EM_Cube-0.800.stl`) so that you can simply drag and drop whatever cubes you want into the slicer, click Arrange, and print. No need to worry about placement or order since they're all labeled.
- **Find [Regular expression]** `^(; printing object EM_Cube-([0-9.]+).*)\.stl.+$`
- **Replace with** `${1}\nM221 S${2}`.

For the record, I am currently using the following setup, enhanced from @jleutgeb1's original post (which is enhanced over the one I came up with in #75 before seeing this post):
```
; Add to Start G-Code, adjust values
; Object 1 will print at the flow rate specified; each subsequent object will have the flow rate increased by the specified increment rate. Set increment to a negative value to go from high to low instead.
{
global flowtestrate = 97;
global increment = 0.5;
"M221 S" + flowtestrate;
}

; Add to Between objects G-code
{
flowtestrate = flowtestrate + increment;
"M221 S" + flowtestrate;
}
```

totalguy, 2023-09-20:
Is there a video on how to do this all this text is very overwhelming and i am really confused. I work much better with visuals text is just too much! ive got the usual issue i cant add an instance of filament on Prusa slicer then extrusion multiplier. 

is there a simple way to do this workaround? everyones talking about code here and there and i just cant get my head around it. ive prusa slicer 2.6.1 and a prusa style printer AM8MU