AndrewEllis93/Print-Tuning-Guide #75: Extrusion Multiplier > Steps > PrusaSlicer - information provided is incorrect, there IS a way, and it's easier than your SS instructions

hiltonjanfield, 2023-05-15:
This should probably be a pull request but I don't have time to go through all that.
I've been working through the print tuning guide today (nice job!) and noticed the following:

> Steps - Slice the test cubes with variations of 1-2% EM
> Prusa Slicer - There is no way to set the EM per object. You will have to print the test objects one at a time.

This is easily accomplished with a PrusaSlicer gcode macro.

Set your EM to 1 in filament settings, as with SuperSlicer.

ctrl-f between_objects_gcode and paste in this (or just the last line of it):
```
; AFTER_OBJECT
; Object [current_object_idx]
; Reduce Extrusion Multiplier by 1 for next object
M221 S{100 - current_object_idx * 1}
```

The first object will print at 100% (as set in the filament settings); each subsequent object will print 1% less. The last number on the last line can be modified to change the decrements - `* 2` will decrease EM by 2% for each object, `* 3` by 3%, etc.

If you really want to go nuts, change your EM to 1.04 or put `M221 S104` at the end of your Start gcode and change the `100` in the last line to `104`. Now your first object will print at 104% EM and decrease from there.

Wood578Guy, 2023-05-31:
Confirmed - this works great...saves a TON of time for Prusaslicer.  Bravo sir!

myevo8u, 2023-06-01:
Just wanted to say, I also created a quick post processing script yesterday for this as well! https://github.com/myevo8u/Prusa-Slicer-Extrusion-Multiplier-Calibration-Script

hiltonjanfield, 2023-06-01:
But who needs external scripts when you can do it with Jinja?

AndrewEllis93, 2023-06-13:
Closing as duplicate of #74, we can continue discussion there
I will likely add one of these methods once I find the time / willpower