AndrewEllis93/Print-Tuning-Guide #77: Extruder Calibration - Step 8 - incorrect Klipper equation

Wood578Guy, 2023-05-31:
Step 8 lists this in the Klipper section:

`<new_rotation_distance> = <previous_rotation_distance> * ( <actual_extrude_distance> / 100 )`

it should be:

`<new_rotation_distance> = <previous_rotation_distance> * ( 100 / <actual_extrude_distance>)`

AndrewEllis93, 2023-07-02:
Klipper documentation has the formula as `rotation_distance = <previous_rotation_distance> * <actual_extrude_distance> / <requested_extrude_distance>`. In our case, 100 is the requested distance, making it  `rotation_distance = <previous_rotation_distance> * <actual_extrude_distance> / 100`, unless I'm missing something here.

Wood578Guy, 2023-07-03:
I was confused when I submitted this...  you are correct - no change needed, docs correct as written.  Sorry for the time you had to spend looking that up.