AndrewEllis93/Print-Tuning-Guide #82: TEST_SPEED Macro Move out of range

seanmccully, 2023-06-19:
Wondering if having to do with sensorless homing, but get an error `Move out of range` when trying to run `TEST_SPEED` macro.

`TEST_SPEED ACCEL=5000 ITERATIONS=2
7:12 PM
TEST_SPEED: starting 2 iterations at speed 350, accel 5000
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.70A Hold Current: 0.70A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Run Current: 0.89A Hold Current: 0.89A
7:12 PM
Move out of range: 689.000 709.000 10.000 [0.000]
7:12 PM
Move out of range: 689.000 709.000 10.000 [0.000]
`

blownupp, 2023-06-21:
This happens to me too:

LDO 2.4r2 350mm

`8:28 PM    

mcu: stepper_x:1741 stepper_y:-1279 stepper_z:-7 stepper_z1:23 stepper_z2:2020 stepper_z3:-6    

stepper: stepper_x:690.003125 stepper_y:-0.000000 stepper_z:15.000000 stepper_z1:15.000000 stepper_z2:15.000000     
stepper_z3:15.000000    

kinematic: X:345.001562 Y:345.001562 Z:15.000000    

toolhead: X:345.001562 Y:345.001562 Z:15.000000 E:0.000000    

gcode: X:345.001562 Y:345.001562 Z:15.000000 E:0.000000    

gcode base: X:0.000000 Y:0.000000 Z:0.000000 E:0.000000    

gcode homing: X:0.000000 Y:0.000000 Z:0.000000    

8:28 PM    

get_position    

8:25 PM    

Move out of range: 699.002 699.002 15.000 [0.000]    

8:25 PM    

Move out of range: 699.002 699.002 15.000 [0.000]`

m-rv, 2023-07-02:
I've had the same issue on my Voron 0 and I've figured out the reason.
The `TEST_SPEED` macro relies on it setting the `G90` in the initial homing sequence. However I was using the `sensorless` homing macros from Voron team and there they are setting `G91`. That means that after the homing macro the coordinates were `printer.toolhead.axis_maximum.x-10` and the script was calling `G1` with `X` and `Y` `printer.toolhead.axis_maximum.x-1` assuming it was still in `G90`, but in `G91` it resulted in almost doubling the coordinates: `2*printer.toolhead.axis_maximum.x-11`

I've added a `SAVE_GCODE_STATE NAME=HOMING` at the start of the homing macros and `RESTORE_GCODE_STATE NAME=HOMING` at the end so that it preserves the `G90/G91` state and it solved the issue

cc @seanmccully @blownupp

SrgntBallistic, 2023-07-04:
> I've added a `SAVE_GCODE_STATE NAME=HOMING`

Would you mind posting your config with those macros? I just tried adding the GCODE_STATE stuff to `_HOME_X`, `_HOME_Y` and `HOME_OVERRIDE` and `homing_override` but still get a grinding collision on X before the macro starts moving.

m-rv, 2023-07-04:
That's my homing macro:
```yaml
[gcode_macro _HOME_X]
gcode:
    SAVE_GCODE_STATE NAME=HOMING
    # Always use consistent run_current on A/B steppers during sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.5 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}

    # Home
    G28 X
    # Move away
    G91
    G1 X-10 F1200
    
    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
    RESTORE_GCODE_STATE NAME=HOMING
```
Note the `G91` code. That cancels the `G90` called in [TEST_SPEED.cfg](https://github.com/AndrewEllis93/Print-Tuning-Guide/blob/6d4afe4020b10b8d045444aea37da93846fe61d0/macros/TEST_SPEED.cfg#L54C12-L54C12):
```yaml
        # Move 50mm away from max position and home again (to help with hall effect endstop accuracy - https://github.com/AndrewEllis93/Print-Tuning-Guide/issues/24)
        G90
        G1 X{printer.toolhead.axis_maximum.x-50} Y{printer.toolhead.axis_maximum.y-50} F{30*60}
```

To test it that's indeed what happens you can try calling the commands from the `TEST_SPEED` macro manually and and checking what happens or inserting additional `G90` and see if it helps

SrgntBallistic, 2023-07-04:
@m-rv 

Here are my homing macros. There a combination of the default skr-pico config and LDO's config.

`[homing_override]
axes: xyz
set_position_z: 0
gcode:
    {% set home_all = 'X' not in params and 'Y' not in params and 'Z' not in params %}
    SAVE_GCODE_STATE NAME=STATE_HOME_OVERRIDE

    # Home Z first, to avoid any situation where the nozzle might scrape the bed
    {% if home_all or 'Z' in params %}
        G28 Z
    {% endif %}

    {% if home_all or 'X' in params %}
        _HOME_X
    {% endif %}

    {% if home_all or 'Y' in params %}
        _HOME_Y
    {% endif %}
    
    G90
    G0 X60 Y60 Z20 F3600 # move to bed center
    RESTORE_GCODE_STATE NAME=STATE_HOME_OVERRIDE`
    
    `[gcode_macro _HOME_X]
gcode:
    SAVE_GCODE_STATE NAME=STATE_HOME_X
    # Always use consistent run_current on A/B steppers during sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.7 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT * RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT * RUN_CURRENT_Y}

    # Home
    G28 X
    # Move away
    G91
    G1 X-10 F1200

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
    RESTORE_GCODE_STATE NAME=STATE_HOME_X

[gcode_macro _HOME_Y]
gcode:
    SAVE_GCODE_STATE NAME=STATE_HOME_Y
    # Set current for sensorless homing
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.7 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT * RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT * RUN_CURRENT_Y}

    # Home
    G28 Y
    # Move away
    G91
    G1 Y-10 F1200

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    # Set current during print
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}
    RESTORE_GCODE_STATE NAME=STATE_HOME_Y

[gcode_macro _HOME_Z]
gcode:
    SAVE_GCODE_STATE NAME=STATE_HOME_Z
    G90
    G28 Z
    G1 Z30
    RESTORE_GCODE_STATE NAME=STATE_HOME_Z`
    
    
Here's a video of what I'm experiencing. Maybe it's not the same issues others are having
https://github.com/AndrewEllis93/Print-Tuning-Guide/assets/25190883/b2d9ccef-fb78-4f01-911f-82aafcf23be6

m-rv, 2023-07-04:
Are you using the TEST_SPEED macro without any modifications?
I don't know if it's breaking anything, but the
```
G90
G0 X60 Y60 Z20 F3600 # move to bed center
```
part in your homing_overwride looks suspicious. And on the side note you have `_HOME_Z` macro you're not using in the overwride, you just do `G28 Z`.
What you have looks similar to what I had, apart from that moving to the center of the bed. Seems like the
`G0 X{printer.toolhead.axis_maximum.x-1} Y{printer.toolhead.axis_maximum.y-1} F{30*60}` line is the one where you get the collision, so you can try adding another `G90` before that? Or going through the commands manually in the console and checking in mainsail interface what the printer thinks is it's current xy position and if it's correct

itsgudenuf, 2023-11-08:
The SAVE_GCODE_STATE NAME=HOMING and RESTORE_GCODE_STATE NAME=HOMING added to my homing macros solved the problem.

skyaee, 2024-03-02:
> SAVE_GCODE_STATE NAME=HOMING and RESTORE_GCODE_STATE NAME=HOMING added to my homing macro solved the problem.

Do you have the complete code for reference?

skyaee, 2024-03-02:
Running tips:Move out of range: 694.000 714.006 0.000 [0.000]

twintersx, 2024-10-14:
Here is my X, Y & Z Homing Macros, adding the additional lines worked right away for me.
`[gcode_macro _HOME_X]
gcode:
    SAVE_GCODE_STATE NAME=HOMING # refer to https://github.com/AndrewEllis93/Print-Tuning-Guide/issues/82
    # Home
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.49 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
    
    SET_KINEMATIC_POSITION X=15
    G91
    G1 X-15 F1200
    
    #G4 P2000
    #M400 to finish all pending moves/process the buffer
    M400 
    G28 X
    
    # Move away
    G91
    G1 X-15 F1200

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000
    
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}

    RESTORE_GCODE_STATE NAME=HOMING

[gcode_macro _HOME_Y]
gcode:
    SAVE_GCODE_STATE NAME=HOMING
    
    {% set RUN_CURRENT_X = printer.configfile.settings['tmc2209 stepper_x'].run_current|float %}
    {% set RUN_CURRENT_Y = printer.configfile.settings['tmc2209 stepper_y'].run_current|float %}
    {% set HOME_CURRENT = 0.49 %}
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CURRENT}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CURRENT}
    
    SET_KINEMATIC_POSITION Y=15
    G91
    G1 Y-15 F1200
    
    #G4 P2000
    #M400 to finish all pending moves/process the buffer
    M400 
    
    # Home
    G28 Y
    # Move away
    G91
    G1 Y-15 F1200

    # Wait just a second… (give StallGuard registers time to clear)
    G4 P1000

    # Wait just a second… (give StallGuard registers time to clear)
    SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CURRENT_X}
    SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CURRENT_Y}

    RESTORE_GCODE_STATE NAME=HOMING

[gcode_macro _HOME_Z]
gcode:
  SAVE_GCODE_STATE NAME=HOMING
  G28 Z
  G90
  RESTORE_GCODE_STATE NAME=HOMING`