AndrewEllis93/Print-Tuning-Guide #85: Expectations with sensorless homing wrt minor skipping with TEST_SPEED

Gallion, 2023-07-09:
I think it would be useful to add information about what can be expected when running this test with sensorless homing. Running a script with just homing, moving, homing, moving etc on my Prusa Bear, I get the following mcu values : 

x : -128, -192, -256, -319, -384
y : -129, -192, -256, -321, -320

The following information is relevant : 

"A stepper motor itself can only lose steps in increments of 4 full steps. (So, if one is using 16 microsteps, then a lost step on the stepper would result in the "mcu:" step counter being off by a multiple of 64 microsteps.)"
[Source](https://github.com/Klipper3d/klipper/issues/1220#issuecomment-462138564) 

Sensorless homing allows to home an axis without the need for a physical limit switch. Instead, the carriage on the axis is moved into the mechanical limit making the stepper motor lose steps. The stepper driver senses the lost steps and indicates this to the controlling MCU (Klipper) by toggling a pin. This information can be used by Klipper as end stop for the axis.
[Source](https://www.klipper3d.org/TMC_Drivers.html?h=sensorless#sensorless-homing)

For these reasons, I don't know if this method of measuring acceleration on printers like this is as useful. There might be a need for a slightly different procedure, maybe one where the acceleration is increased until audible skipping happens, then rolled back by a certain amount. Or perhaps a helpful method would be to print a tower, executing the test movements every n layers, then seeing if there is any visible shift.

Hun73rdk, 2023-10-06:
Yea i just set my new voron 2.4 with with sensorless a and at a acceleration of 500 and travel speed at 300 it get numbers far apart.

mcu: stepper_x:61549 stepper_y:-43787 stepper_z:172
mcu: stepper_x:61793 stepper_y:-43757 stepper_z:163

Must be a way to compensate for that but i have no clue

l-ejs-l, 2023-12-21:
i have the exact same issue, no idea how to determine if i've lost a step with sensorless homing.

nikoramiro, 2024-04-29:
I'm unsure if my logic is sound, but the way I ended up doing this is by expecting my axis to skip a full step when homing (so 64 microsteps in my case), if the numbers differed more or less than 64 I assumed minor skipping had occured and dialed down my values.