AndrewEllis93/Print-Tuning-Guide #93: Suggestion for Tutorial Page

PuterPro, 2023-08-28:
In the print surface prep you suggest:

"Smooth PEI
Consider scuffing it up with a :page_facing_up:[ kitchen scouring pad](https://www.scotch-brite.com/3M/en_US/scotch-brite/tools/~/Scotch-Brite-Heavy-Duty-Scour-Pad/?N=4337+3294529207+3294631680&rt=rud)."

I would suggest you add that for ALL smooth surfaces a White Melamine sponge (ie: Mr Clean pad or clones) does a _great_ job of cleaning heavily contaminated surfaces easily, and also give a more mild "tooth" than the green scrub pads. This retains the shiny finish a smooth plate gives and still increases adhesion. Win-Win.

alanmakestuff, 2023-09-13:
I also have additional suggestions that have helped immensely with surface prep.

For Smooth PEI;

- Consider scuffing the smooth surface once in a while with a Diapad (diamond dust coated foam pad used in glass polishing). Use circular motions. The scratches are cleaner and provide better adhesion than sandpaper. They are a bit pricey but you only need one Diapad for a lifetime.

- To clean I would suggest WD-40 Specialist Degreaser spray instead of isopropanol. The adhesion obtained never fails to disappoint. [https://www.wd40.com/products/industrial_strength_degreaser/](url)

PuterPro, 2023-09-13:
> * Consider scuffing the smooth surface once in a while with a Diapad (diamond dust coated foam pad used in glass polishing). 
GREAT TIP! But ... These are really hard to find (cheaply). Fortunately I have a set from years ago, still in good shape, so not a problem. But it's funny what pops up if you search Diapad on DuckDuckGo - it takes you Dialpad - a Marketing app! LOL Amazon's even funnier - they thought I was searching for adult DIAPERS! {Crying}

> * To clean I would suggest WD-40 Specialist Degreaser spray instead of isopropanol. The adhesion obtained never fails to disappoint. [https://www.wd40.com/products/industrial_strength_degreaser/](url)

The WD40 Degreaser is great too, I didn't know they made that, it will be good for LOTS of stuff. Thanks!!