AndrewEllis93/Print-Tuning-Guide #114: Ellis' Print Tuning Guide compatibility?

cwadej, 2024-02-11:
Is the guide compatible with Ender-3 V3 KE ?
First step of calibration "Open your printer.cfg and find the [extruder] section."   is this in the printer software, or???
I find nothign like it in the printer or the creality workbench app linked to the printer

pvginkel, 2024-03-06:
I have the same question. I'm trying G-Code commands from the manual and I find that they're not working, specifically this one at the beginning of the Extruder Calibration tutorial:

```
G1 E1 F60
```

It gives a E500 key:2111 error. Is it really a requirement to install the Klipper firmware to be able to run the guide?

clearbucketLabs, 2024-03-06:
> I have the same question. I'm trying G-Code commands from the manual and I find that they're not working, specifically this one at the beginning of the Extruder Calibration tutorial:
>
> ```
> G1 E1 F60 [...]

well you didn't pre-heat the nozzle.... if you're extruding cold without the hotend attached you'll have to look up commands to disable that check.

However the guide clearly only has klipper and reprapfirmware... some things will work but firmware specific commands wont, you'll have to look them equivalents for your machine.

pvginkel, 2024-03-06:
I'm very new to this. I have the KE, which should be Klipper, but I'm realizing it's really not the same thing as normal Klipper firmware. You're right about the temperature. After I add that command, the extrude command works properly. However, I'm now finding that I have no way to read data back by just executing G-Code files (the `M503` command), and I'm not able to get the USB serial connection to work on Windows. You're right, there's no use attempting these parts of the guide without the normal Klipper firmware.

clearbucketLabs, 2024-03-06:
> I'm very new to this. I have the KE, which should be Klipper, but I'm realizing it's really not the same thing as normal Klipper firmware. You're right about the temperature. After I add that [...]

it's probably not the firmware, Klipper doesn't have a M503 gcode....