AndrewEllis93/Print-Tuning-Guide #117: Question about SuperSlicer version

Terixxx, 2024-02-27:
Hello, 
In provided profiles is "Required SuperSlicer version: [📄2.4.58.5](https://github.com/supermerill/SuperSlicer/releases/tag/2.4.58.5)" but I wonder if I update the SuperSlicer to the latest version? Is it mandatory to use 2.4.58.5?

bassamanator, 2024-03-10:
Best to use original version (old official release) unless you're having issues with it.

gazialankus, 2024-03-31:
Btw SuperSlicer seems to have stopped development a while ago. It does not even have tree supports. https://github.com/supermerill/SuperSlicer/discussions/1944

bassamanator, 2024-03-31:
> Btw SuperSlicer seems to have stopped development a while ago. It does not even have tree supports. [supermerill/SuperSlicer#1944](https://github.com/supermerill/SuperSlicer/discussions/1944)

Merill is doing SuperSlicer full-time now. Besides a few pre-releases that were released this year, he said it's going to be a merge with PrusaSlicer 2.7.x next.