AndrewEllis93/Print-Tuning-Guide #121: There is an Error in the calculation of the new_rotation_distance in the Tuning Chapter Extruder Calibration

iTzMtrx, 2024-03-20:
[image]
Hey, I think there must be an error in the calculation of the new rotation distance. There is a difference between the calculations of Klipper Marlin and RepRapFirmware. I dont know if this is intended, but I just wanted to make sure, that it is correct or gets corrected.

brymck1, 2025-03-21:
Yeah definitely still needs correcting, I am in the process of calibrating klipper and noted it as well...

dowatugkins, 2025-05-08:
The math should be `actual_distance/expected_distance`. You are looking for a percentage of travel. You expected it to travel 100mm but got x instead. If x is say 101mm, 101/100 =1.01, or 101%, or if x is 99mm, 99/100=0.99 or 99%. The reprap and marlin formulas are wrong.