AndrewEllis93/Print-Tuning-Guide #142: TEST_SPEED: "Move out of range: 356.000 350.000 60.000 [0.000]"

Sprocyon, 2024-10-13:
Hi, I'm not getting the "TEST_SPEED:"script to work for some reason. "Move out of range: 356.000 350.000 60.000 [0.000]" Most likely the point is that I have a sensor less homing, and there is something in the variables, but I am not able to understand it. Can someone tell me how to run this script more easily without mastering all the nuances of writing scripts :)

gamkiller77, 2025-02-26:
I am having the same problem. Mine is giving a range of 500, 500 something not sure where it pulling that from.  I am also running sensor less homing.

batista16, 2025-04-01:
Found the issue, after every G28, I need a G90.. not sure if that has something to do with my sensorless homing macro.. but it could be.. anyway the macro could have it just for it to work even if this is the case.. it's missing after line 71.